robincornelius/libedssharp #138: Export canopennode c/h menu entry file name can't be changed

martinwag, 2018-04-26:
Hi, 

when selecting "export CanOpenNode c/h" menu entry, I can change the file name in the dialog box. However, this change is ignored when saving the files.

Should we change this behaviour or just make it non-modifiable?

robincornelius, 2019-01-30:
I'll revisit this one at some point. I wanted to use a FolderBrowserDialog but its horrible, almost no one uses it as no one likes it and everyone uses a file dialog instead. But yes its confusing if you then change the file name. Something should happen or  warning or something...

CaCtus491, 2019-05-25:
Just another perspective on this... We have common code base across several devices that use CANopenNode. Currently we generate the CO_OD.[c,h] files and rename them manually. The ability to change the name of the output file would be a welcome one in our case.

We just have a few #ifdef statements add to CANopen.h to select the appropriate one for the current build.

trojanobelix, 2019-10-02:
I also need to save the files with different file names. I have tried to fix it. Get my pull request and give it a try, if you want.
For me it works fine so far.

robincornelius, 2019-10-22:
Fixed at bbf5a85bb24f00c758c659ebbb811ed21fef2bc7
Thanks trojanobleix

closed 2019-10-22 by robincornelius