robincornelius/libedssharp #139: Export XDD file - Memory Types are not stored

martinwag, 2018-04-26:
Hi,

when exporting a CANopen xml to XDD the memory types seem to be lost. When importing a XDD file, the memory types seem to be assinged at random (?).

martinwag, 2018-04-26:
This seems to apply also for the OD entries marked as not "Enabled".
What are your plans regarding XDD? Should this replace the CANopenNode xml format or remain an additional export/import option?
As far as I can see, my CO_OD.c/h otherwise matches from both file formats.

robincornelius, 2019-01-30:
xdd ideally should replace it as its the CanOpen offical standard, but i'll need to take a look to see if there are any easy ways to "extend" it with these extra attributes.

robincornelius, 2019-02-01:
I've added memory type support to XDD, so now its pretty complete. I believe the main file format for the app should be XDD (for individual files anyway) I'm going to change it to default save to XDD, but Save As/Export .. all the other formats. I'll probably create an array of XDD containers to make the network format also work in XDD. But i'm not going to remove importing/exporting any of the old formats there is too much old stuff using them. I've got loads my self and i'm sure everyone else does to

commit f4e2b116 (2019-02-01): Add extra attributes to the XDD to permit storing of the storage location with in the XDD format Fixes #139. It appears there are no usable extension attributes or other fields that can be used so i've just added a new attribute. This *may* create a warning in other editors when loading a file as this is officially an unsupported attribute. But really they should just ignore it.
closed 2019-02-01 by robincornelius