robincornelius/libedssharp #147: Compiling canopend failed with CO_OD.h & CO_OD.c generated with libedssharp

gdherbom, 2018-11-29:
Hello, 

First, thanks for the application. 
I get some compilation errors of canopend when generating CO_OD.c file with a .eds file provided by a device provider so I tried with the CanOpenSocket.eds of CANopenNode/CANopenSocket project (https://github.com/CANopenNode/CANopenSocket/blob/master/canopend/objDict/CANopenSocket.eds). 

1. the CO_OD.c and CO_OD.h files generated by libedssharp are not the same as those in canopend project. 
2. when I try to build canopend, I get some link errors: 
```
cc -Wall -g -I../CANopenNode/stack/socketCAN -I../CANopenNode/stack -I../CANopenNode -I./src -I./objDict -I./app -c app/application.c -o app/application.o
cc -lrt -pthread ../CANopenNode/stack/socketCAN/CO_driver.o ../CANopenNode/stack/socketCAN/CO_OD_storage.o ../CANopenNode/stack/socketCAN/CO_Linux_tasks.o ../CANopenNode/stack/crc16-ccitt.o ../CANopenNode/stack/CO_SDO.o ../CANopenNode/stack/CO_Emergency.o ../CANopenNode/stack/CO_NMT_Heartbeat.o ../CANopenNode/stack/CO_SYNC.o ../CANopenNode/stack/CO_PDO.o ../CANopenNode/stack/CO_HBconsumer.o ../CANopenNode/stack/CO_SDOmaster.o ../CANopenNode/stack/CO_trace.o ../CANopenNode/CANopen.o src/CO_command.o src/CO_comm_helpers.o src/CO_master.o src/CO_time.o src/main.o objDict/CO_OD.o app/application.o -o app/canopend
src/CO_command.o: In function `command_process':
/home/dev/CANopenSocket/canopend/src/CO_command.c:423: undefined reference to `CO_sendNMTcommand'
/home/dev/CANopenSocket/canopend/src/CO_command.c:436: undefined reference to `CO_sendNMTcommand'
/home/dev/CANopenSocket/canopend/src/CO_command.c:449: undefined reference to `CO_sendNMTcommand'
/home/dev/CANopenSocket/canopend/src/CO_command.c:466: undefined reference to `CO_sendNMTcommand'
/home/dev/CANopenSocket/canopend/src/CO_command.c:473: undefined reference to `CO_sendNMTcommand'
collect2: error: ld returned 1 exit status
Makefile:72: recipe for target 'app/canopend' failed
make: *** [app/canopend] Error 1

```

Please could you check if you have the same problem or where I am wrong? 

Thank you. 
Gerald

robincornelius, 2018-11-29:
Hi,

CO_sendNMTcommand is hidden by an include guard that is only turned on if

    #define CO_NO_NMT_MASTER 1

is present in the project (usually CO_OD.h), in order to get the ODE
to export with this set to 1 you must add object 0x1f80 to the object
dictionary and set as a minimum bit 0 to 1 (eg set value to 0x01)

Object 0x1f80 bit (0) defines if a node is a NMT master

If this  bit is set, the define is exported to allow the device to act as
NMT master.

Hope this helps

Robin

gdherbom, 2018-11-30:
Hello Robin, 

Thank you for the explanation. It works now! I can compile and run. 
However I am not very experimented with CanOpen protocol and I think I am wrong somewhere when I configure the PDO mapping. 

My goal is to read data coming from a Joystick. The provider of this equipment gives me the following table: 
[image]

I don't really understand how to map the CAN ID (0x185) to the correct RPDO. When I test my code with canopend, I get always the value 0. However the Joystick emitted  correctly: 

```
$ candump can0  
  can0  185   [8]  00 00 00 00 00 00 81 CA
  can0  185   [8]  00 00 00 00 00 00 82 C3
  can0  185   [8]  00 00 00 00 00 00 83 C4
  can0  185   [8]  00 00 00 00 00 00 84 D1
  can0  185   [8]  00 00 00 00 00 00 85 D6
  can0  185   [8]  00 00 00 00 00 00 86 DF
  can0  185   [8]  00 00 00 00 00 00 87 D8
  can0  185   [8]  00 00 00 00 00 00 88 F5
  can0  185   [8]  00 00 00 00 00 00 89 F2
  can0  185   [8]  00 00 00 00 00 00 8A FB
  can0  185   [8]  00 00 00 00 00 00 8B FC 
```
If you can just explain me how to configure the RPDO mapping of checksum (last column) with your application, it would be very nice ! 

I am reading values in the application.c file of canopend with this code: 
```
void app_program1ms(void)
{    
    for (int i = 0 ; i <= 7 ; ++i)
    {
        printf("PDO1 %d 0x%02x -- ",i, CO_OD_RAM.receivePDO1MappingData[i]);
        printf("PDO2 %d 0x%02x -- ",i, CO_OD_RAM.receivePDO2MappingData[i]);
        printf("PDO3 %d 0x%02x -- ",i, CO_OD_RAM.receivePDO3MappingData[i]);
        printf("PDO4 %d 0x%02x -- ",i, CO_OD_RAM.receivePDO4MappingData[i]);
    }
    printf ("\n\n");
}
```

Thank you. 
Gerald

JuPrgn, 2018-11-30:
Hi Gerald,

I suggest you to download CiA DS 301 PDOs are very well explained.

On your EDS you have to create an SDO object for your RPDO datas, for example : 

0x6200 : OD_PDOdata

Then you have to configure your RPDO : 

0x1400 : RPDO Communication Parameter - here you configure your PDO parameters (COB-ID : 0x180, type, inhibit time...)
0x1400 subindex 1 : 0x180  COB-ID 0x180 + Node_ID
0x1400 subindex 2 : 0xFF, asynchronous PDO
...

0x1600 : RPDO Mapping Parameter - here you configure how many bits you will map and where you want them to be mapped.
0x1600 subindex 1 : 0x62000108 : address 0x6200, subindex 01, copy 08 bits
and on the others subindex you can map other bits to other locations.

On your application you can read this PDO using OD_PDOdata : 

`printf("PDO1 %d 0x%02x -- ",i, OD_PDOdata);`

Julien

robincornelius, 2018-11-30:
In the ODE,  open your starting eds/xml file. On the object dictionary tab,
on manufacture objects, right click and "Add New Object"

Set it to the following:-

Index: 0x2000
name: joystick
objecttype: array
Data Type: UNSIGNED8
Subindexes: 8

Press "Create"

On the right select PDO Map and set it to "Optional" press SAVE

Goto RX PDO Tab

Either edit an existing PDO or create a new one, for each byte 1-8 use the
dropdown boxes to select 0x2000/01/ 0x2000/02 ...

Ensure the COB is set to 0x185 and hit save (you may need to click on the
bit of the table that says "Assigned PDO mappings" to be able to edit COB,
make sure $NODEID is gone

At this point your RX PDO is mapped and set up now in your application you
should be able to use the variable OD_joystick[..] and that is just a
regular uint8_t [8] array that exactly matches the array the data is
transmitted in

Hope this helps

Robin

gdherbom, 2018-12-11:
Hi Julien and Robin,

Thank you for your help. I can now well understand the working of PDO.
However, my tests don't work and I don't understand why. I follow the guidelines provided by Robin.

Here is a screenshot of the OD definition : 
[image]

And the PDO mapping:
[image]

**I have some doubt about the type in communication parameters so I let the default value (255)**. 

It correctly generates the CO_OD.c & codes CO_OD.h, building is ok but data are not copied in OD_joystick variable. 

Here is in the first terminal the output of canopend and in the bottom terminal the result of candump command : 
[image]

Do I need to write some initialization codes in my canopend app to enable the PDO mapping? 

Thanks again. 
Gerald

JuPrgn, 2018-12-11:
Hi Gerald

Did you enter in operational mode on your receiving device ?

The PDO Rx and Tx are processed in NMT operational mode only. 

Julien

robincornelius, 2018-12-11:
I see from the log :-

701 [1] 7F  and this is a pre-operation heartbeat

So your device is in pre-op mode. this is a bit odd as the OP set object
0x1f80 to 0x01
which should set the device as an NMT master and automatically enter
operational at startup.

So next question is why is it in pre-op? its possible if something else has
gone wrong there was an emergency code for node 1 0x81 .... right after the
boot message 0x701 [1] 00 and if so the device will drop into preop

Robin

robincornelius, 2019-01-30:
The original issue with the ODE was fixed so closing the issue.

closed 2019-01-30 by robincornelius