robincornelius/libedssharp #149: Why is a VISIBLE_STRING is a CO_ODA_MB_VALUE

rubenverhoef, 2019-01-25:
Why is a VISIBLE_STRING an MultiByte value? it is an array of bytes with length x.
A VISIBLE_STRING gets atribute 0x80 but it should not have that attribute i think:
https://github.com/robincornelius/libedssharp/blob/e80939b4e4b82a1c745f7ed6505fec5c76d1a2d7/libEDSsharp/CanOpenNodeExporter.cs#L800

robincornelius, 2019-01-27:
If i understand the way can open node works correctly, any value that is
not a single byte needs this set as for any values that consist of multiple
bytes (this includes uint16 unit32 etc).because of the BIG/LITTLE endian
conversion that is in the code

If you look at the code in CO_SDO.c you will find :-

 /* swap data if processor is not little endian (CANopen is) */
#ifdef CO_BIG_ENDIAN
    if((SDO->ODF_arg.attribute & CO_ODA_MB_VALUE) != 0){
     ...
 so if the CO_BIG_ENDIAN is defined AND its a multibyte (params has 0x80
set) then the endian conversion must be run to ensure correct byte order on
the wire of any SDO/PDO transfer

*So i think its correct and required

Regards

Robin

rubenverhoef, 2019-01-28:
https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/example/CO_OD.c#L310

rubenverhoef, 2019-01-28:
In the CO_OD example from CANopenNode a VISIBLE_STRING is not a Multi Byte value, see:

https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/example/CO_OD.c#L310
And:
https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/example/CO_OD.h#L237

I don't know how they do the conversion (Big/Little) with the array of bytes and if it is needed.

My MCU is 16Bit oriented so my arrays are 16bit arrays. For this reason i have to know if it is a array of bytes with length x or it is a MB_VALUE(16bit/32bit) with x length of  bytes.

My code works correctly with the example .c/.h file.

robincornelius, 2019-01-28:
On Mon, 28 Jan 2019 at 07:34, rubenverhoef <notifications@github.com> wrote:

> In the CO_OD example from CANopenNode a VISIBLE_STRING is not a Multi Byte
> value, see:
>

Yes you are correct, i've re-read DS-301 and a VISIBLE_STRING is an array
of VISIBLE_CHAR so is NOT multibyte. I suspect the same bug exists for
OCTET_STRING as well as that is also just an array (UNSIGNED8),
UNICODE_STRING however is an array of UNSIGNED16 so that is multibyte. I
think these are the only issues under DS301 9.6.1 Extended Data Types.

I'll patch this later today and upload to the XDD branch.

Thanks for spotting this

Regards

Robin

rubenverhoef, 2019-01-28:
Okay, great!
Thanks for the quick respons.

rubenverhoef, 2019-01-29:
When you have time, can you also make an new installer? And maybe with the #150 fix (if you agree with the problem)?
Thanks a lot

robincornelius, 2019-01-30:
Hi,

Yes i'm planning on this. I wanted to review any other outstanding issues
and then package them all at once.

Regards

Robin

robincornelius, 2019-01-30:
As this is fixed in git and i'm releasing soon

commit 875fc9ce (2019-01-28): Don't set multibyte flag on VISIBLE_STRING fixes #149
closed 2019-01-30 by robincornelius