robincornelius/libedssharp #154: Command button not visable

trojanobelix, 2019-02-01:
[image]

Hi Robin,

I am trying the 0.7 on the Xdd branch. I have the problem, that I can not see the command button to save changes in the OD .  I use VS2017 to compile the source code.

robincornelius, 2019-02-01:
What screen size are you using?

Ah i know whats happened, the extra field as pushed it just out of reach at
the bottom. I'll fiddle with the layout

robincornelius, 2019-02-01:
AH no its not that, on mine i can shrink/grow the window and the
description field will change size and the  subindex list box changes size
as well and it all fits any size down to quite a small display. Wonder why
its working differently for you.. if you resize the window (drag it around)
does it fix itsself?

trojanobelix, 2019-02-01:
Nothing special. 1920x1080. External IIYAMA monitor on a HP notebook. 

Microsoft Windows 7 Professional 	6.1.7601 Service Pack 1 Build 7601
Hewlett-Packard HP ProBook 6570b 
Intel(R) Core(TM) i5-3210M CPU @ 2.50GHz, 2501 MHz, 2 Kern(e), 4 logische(r) Prozessor(en)
BIOS	Hewlett-Packard 68ICE Ver. F.42, 20.05.2013

trojanobelix, 2019-02-01:
No. I can resize and maximize, but the button is always mostly  pushed it just out of reach. Always on the position you see on the picture.

trojanobelix, 2019-02-01:
Small 
[image]

Maximized
[image]

robincornelius, 2019-02-01:
> Not sure, but I think it changed with my change to the XDD branch some
> month ago.
> I just "use" the software and forgot to report it.

I've put the controls in a container and adjusted the anchor, see if that
helps

trojanobelix, 2019-02-01:
Okay. The button is visible now.
But I do not get a warning when leaving the window with data changed. The button does not change to red background on data change.

robincornelius, 2019-02-01:
Ah yea, putting the controls inside a group box caused them to not be added
to the changed notification. Should be fixed.