robincornelius/libedssharp #156: CANeds warnings

trojanobelix, 2019-02-05:
Hi Robin,
using CANeds from Vector gives a lot of warning. Nothing really serious, but this one might be corrected:
**warning 1000: Object [1016sub0]: Access type differs from selected database  (value: RO).**
That means the sub 0 (max sub-index)  should be RO

Really often you get a "Data type differs from selected database  (no value allowed)."
because DS301 also knows an empty field, that seem not be supported by the OD  editor.

And the 1F81/2sub1-7F gives a  "Parameter name is missing".

Again, nothing serious, but maybe some polish.

btw: If I save the XDD under  CANeds I can not open it in the OD editor. There seems to be a non-compatibility. But it might be a Vector problem 

Best regards.

Load C:\Git_lokal\AZR_DSP\Main\CANopenApp\alcosmartOD\IPA_system_slave.xdd ...
Compare read EDS with database...
warning 1000: Object [1003]: Data type differs from selected database  (no value allowed).
warning 1000: Object [1003]: Access type differs from selected database  (no value allowed).
warning 1000: Object [100C]: Default value differs from selected database  (value: 0).
warning 1000: Object [100D]: Default value differs from selected database  (value: 0x00).
warning 1000: Object [1010]: Data type differs from selected database  (no value allowed).
warning 1000: Object [1010]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1011]: Data type differs from selected database  (no value allowed).
warning 1000: Object [1011]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1012]: Access type differs from selected database  (value: RW).
warning 1000: Object [1015]: Default value differs from selected database  (value: 0x0).
warning 1000: Object [1016]: Data type differs from selected database  (no value allowed).
warning 1000: Object [1016]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1016sub0]: Access type differs from selected database  (value: RO).
warning 1000: Object [1017]: Default value differs from selected database  (value: 0).
warning 1000: Object [1018]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1029]: Data type differs from selected database  (no value allowed).
warning 1000: Object [1029]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1200]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1280]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1400]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1401]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1402]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1403]: Data type differs from selected database  (no value allowed).
warning 1000: Object [1403]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1600]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1601]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1602]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1603]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1800]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1800sub5]: Default value differs from selected database  (value: 0).
warning 1000: Object [1801]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1801sub5]: Default value differs from selected database  (value: 0).
warning 1000: Object [1802]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1802sub5]: Default value differs from selected database  (value: 0).
warning 1000: Object [1803]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1A00]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1A01]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1A02]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1A03]: Access type differs from selected database  (no value allowed).
warning 1000: Object [1F81]: Data type differs from selected database  (no value allowed).
warning 1000: Object [1F81]: Access type differs from selected database  (no value allowed).
warning 1003: Object [1F81sub0]: Parameter name is missing.
warning 1003: Object [1F81sub1]: Parameter name is missing.
...
warning 1003: Object [1F81sub7F]: Parameter name is missing.
warning 1000: Object [1F82]: Data type differs from selected database  (no value allowed).
warning 1000: Object [1F82]: Access type differs from selected database  (no value allowed).
warning 1001: Object [1F82]: Number of sub objects is too small  (value: 129).
warning 1003: Object [1F82sub0]: Parameter name is missing.
warning 1000: Object [1F82sub0]: Default value differs from selected database  (value: 128).
warning 1003: Object [1F82sub1]: Parameter name is missing.
...
warning 1003: Object [1F82sub7F]: Parameter name is missing.
End of comparison.

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Needs external interoperability fixture/tool: Vector CANeds validation unavailable.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Keeping this issue open because the remaining feature, compatibility decision or reproduction requirement is not resolved by this change.

trojanobelix, 2026-10-20:
Hi robincornelius, nice to "see" you again.

Here is the link for anyone interested in trying out the CANeds checker. [https://www.vector.com/de/services/download-center/#types:freeware&product:caneds](https://www.vector.com/de/services/download-center/#types:freeware&product:caneds)

closed 2026-10-20 by trojanobelix

robincornelius, 2026-10-20:
> 2\. Hi robincornelius, nice to "see" you again.
>     Here is the link for anyone interested in trying out the CANeds checker. https://www.vector.com/de/services/download-center/#types:freeware&product:caneds

Hi,

Sorry been very busy with other things and little time for open source projects. But managing to make some progress on some now.

Thanks for the link, will try and have a review of that and get the eds to fully validate.