robincornelius/libedssharp #158: $NODEID is not saved in OD c file

trojanobelix, 2019-02-11:
If I try to export c file with PDO mapping an use the $NODEID define, it is not saved in the c file
 e.g. if the objects COB-ID used by RPDO (0x1400) is set to 0x00000180+$NODEID
in the entry for 0x1400 is 
/*1400*/ {{0x2L, 0x0180L, 0xfeL},

expected is something like /*1400*/ {{0x2L, 0x0180L+$NODEID, 0xfeL},

taxilof, 2019-02-12:
The $NODEID ist not define for the *.c file, but for the *.eds file.

CANopenNode calculates the COB-IDs based on the node ID configured in the code.

trojanobelix, 2019-02-12:
So ther is no difference between 

0x00000180+$NODEID

AminGholamzadeh, 2019-08-15:
@trojanobelix  You're correct. I think the fix should really be in CANopenNode as well as libedssharp. 

In CANopenNode, CO_PDO.C, the functions "CO_RPDOconfigCom" and "CO_TPDOconfigCom" will add the nodeId to the COB-ID for the first 4 TPDO/RPDOs. This is hard-coded. The rest of the PDOs are however left at whatever default value you put in the GUI (+$NODEID will have no effect).

So if you want to add "NodeID" then use first 4 PDOs, otherwise leave first 4 PDOs unused and start from 5th.

robincornelius, 2019-08-15:
Be careful with $NODEID

The $NODEID is used in the editor, it takes what ever value is in the
Concrete Node Id then will substitue this when generating the OD for
occurances of $NODEID and then attempt to add them together. (if this fails
it just uses the inital value)

Also $NODEID is valid in EDS files, the problem with this is the eds file
itself has no concept of what the actual node id is so unless you can pass
this intention through to the stack then it is added at runtime, it is
useless. So the workaround in the editor is to just use what ever is in
concrete node id and apply that (and ofcause the default cobs 0x180 0x200
etc will get node ID added to them by canopen node).

Robin

AminGholamzadeh, 2019-08-15:
Yeah the problem is CANopenNode has no concept of "+$NODEID "
I get my NodeID from DIP switches and so there's no way libedssharp can substitute the $NODEID with the correct value. If CANopenNode implement this then at run-time, just before initialising CO object, it will substitute $NODEID.

All in all there's nothing more libedssharp can do about this I believe so thanks Robin, saved me a lot of nervous coding.

robincornelius, 2019-08-15:
On Tue, 12 Feb 2019 at 10:34, trojanobelix <notifications@github.com> wrote:

> So ther is no difference between
>
> 0x00000180+$NODEID
>

CanopenNode will take any "default cob id"

There is a small difference in the editor if you use the Network PDO
report, currently if you write only a default COB eg 0x00000180 the Network
report will show that, if the PDO is written as 0x00000180+$NODEID then it
will add the concrete NodeID in the report window.

As for what is written to the OD, no it makes no difference 0x00000180
always implies 0x00000180+$NODEID (and it will use the actual node id at
runtime) so if you store that in an eeprom and change it, it will be fine.

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Support question: predefined NODEID values depend on stack/runtime configuration; no new defect established.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Keeping this issue open because the remaining feature, compatibility decision or reproduction requirement is not resolved by this change.