robincornelius/libedssharp #161: Fails to open a particular EDS file

starthal, 2019-05-09:
Trace:

```
System.Collections.Generic.KeyNotFoundException: The given key was not present in the dictionary.
  at System.Collections.Generic.SortedDictionary`2[TKey,TValue].get_Item (System.Collections.Generic.TKey key) <0x40440180 + 0x000d3> in <filename unknown>:0 
  at libEDSsharp.EDSsharp.ParseEDSentry (KeyValuePair`2 kvp) <0x4043e310 + 0x00a53> in <filename unknown>:0 
  at libEDSsharp.EDSsharp.Loadfile (System.String filename) <0x40436310 + 0x0032b> in <filename unknown>:0 
  at ODEditor.ODEditor_MainForm.openEDSfile (System.String path, Filetype ft) <0x404329e0 + 0x00083> in <filename unknown>:0 
```

This looks similar to #145.

The EDS file is one provided by ADVANCED Motion Controls: 

https://info.a-m-c.com/acton/attachment/33817/u-0006/0/-/-/-/-/

Attached here: 
[AMC_Digiflex_1.0.9.zip](https://github.com/robincornelius/libedssharp/files/3162715/AMC_Digiflex_1.0.9.zip)

robincornelius, 2019-05-09:
Thanks for sending the eds file.

By the looks of it there is an error in the file, on line #2502 add this
bit its missing:-

[2022]
ParameterName=Analog Input
ObjectType=0x9
SubNumber=5

insert that just before [2022sub0] its defining subobjects with out a
parent object so the eds file is at fault. It would be more helpful if the
editor gave a warning and a line number in the eds file so may be i can add
that at some point. There is something else going on with the PDO mapping
so take care if you are using the RX/TX pdo tabs in the application as its
unhappy about something when its parsing it.

starthal, 2019-05-09:
Thanks, that was the problem.

I agree the warning feature would be very valuable! But I'll close this issue.

The PDO mapping issue turned out to be a size mismatch (mapping specified a 32-bit length but the object is `Integer64`).

Thanks for making libedssharp!

closed 2019-05-09 by starthal