robincornelius/libedssharp #162: CO_OD.h output for an array of OCTET_STRING

CaCtus491, 2019-05-24:
When I create an array of 4 octet strings, each set with a default value of `00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00` at index 0x6100, I get the following in the CO_OD.h output:
`/*6100      */ OCTET_STRING    CO_DynamicTextString[16][4];`

...it seems that the array dimensions 16 and 4 are swapped around the wrong way. The initialiser in CO_OD.c appears to be correct:
`/*6100*/ {{0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L}, {0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L}, {0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L}, {0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L, 0x0L}},`

CaCtus491, 2019-05-27:
Further to this, in the CO_OD.c output, it specifies the length as 0 (fourth element in the initialiser below when defining `const CO_OD_entry_t CO_OD[]` at the end of the file). The end result is that when I attempt an SDO Download with a client, the server returns an abort code because the length of the supplied data is always greater than the length of the OD entry size (zero).

`
{0x6100, 0x04, 0x0e, 0, (void*)&CO_OD_RAM.CO_DynamicTextString[0]},
};
`

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Fixed: octet-array storage and OD lengths agree.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius