robincornelius/libedssharp #165: Support for Complext Data types (DEFSTRUCTS)

AminGholamzadeh, 2019-09-17:
*Enhancement*

Hi Robin, 

Is there currently any way to implement multiple indexes being records of the same complex type?  

Since the *.c and *.h files will have a structure defined for each index of type "Record", could this be expanded so that multiple indexes of same object could be achieved? (Something like defstructs in original openCAN spec)

At the moment the only way to do this is to have records of different name with the same structure which will create identical "structs" in C with different names (not end of the world, but wasteful)

heliochronix, 2020-02-12:
It would be nice to have the ability to define data types per application. So, for example, have all the standard data types included, but allow selection of (or custom definition of) other data types to include and to have as an option for the data type drop down.

As an example, on one of the projects I work on the TIME_OF_DAY datatype is not useful or used (but defined by default anyway). However, data types based on CCSDS Time Code Formats would be useful. It'd be nice to be able to select which ones I'd like to include, and make my own if necessary.

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Enhancement: DEFSTRUCT data type support remains unimplemented.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Keeping this issue open because the remaining feature, compatibility decision or reproduction requirement is not resolved by this change.