robincornelius/libedssharp #175: export canopennode object dictionary indexes

ArminBmrt, 2019-10-28:
I am having problems exporting eds to CANopenNode .h/.c files. I managed to replicate the problem using the files in CANopenNode repository commit 3f1a636.

What I do is import the io.eds file "CANopenNode/example/IO.eds" and then export the canopennode files. When I try to compile these files I get a lot undeclared errors. For example: 'OD_NMTStartup' undeclared. 

When looking for this index (0x1f80) I see in CO_OD.h:
`#define OD_1f80_NMT_Startup                                 0x1f80`

If I compare this to CO_OD.h file in CANopenNode, " CANopenNode/example/CO_OD.h " is see:
`define OD_NMTStartup                              CO_OD_ROM.NMTStartup`

This is the object name without its index. This is the name CANopenNode is expecting.

robincornelius, 2019-10-28:
Just did a clean test and its working for me *except* i have to disable the trace OD entries from IO.eds. with the trace entries left in it somehow generates an array that is the wrong size. With them removed i can compile the entire stack with CO_OD.c and its all fine. Is it possible that you have the same error and this is then generating a bunch of compiler errors that are hiding the original issue.

ArminBmrt, 2019-10-29:
Thanks for the quick reply. It compiles. I am unable to recreate the problem at the moment. So I will close this issue. Thanks

closed 2019-10-29 by ArminBmrt