robincornelius/libedssharp #176: Faulty code generated if memory type is "ROM" in OD object

trojanobelix, 2019-10-28:
I have added some PDOs and the corresponding 0x160x mapping parameters were set to memory type ROM (seems to be default).
The generated code is not valid. 
If I change the memory type to RAM it compiles without error. 

[image]

`
/***** Definition for ROM variables ********************************************/
struct sCO_OD_ROM CO_OD_ROM = {
           CO_OD_FIRST_LAST_WORD,

/*1403*/ {0x2L, 0x0488L, 0xfeL},
/*1603*/ {0x0L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0x0000L},
/*1802*/ {0x6L, 0x0408L, 0xfeL, 0x00, 0x0L, 0x00, 0x0L},
/*1803*/ {0x6L, 0x0508L, 0xfeL, 0x00, 0x0L, 0x00, 0x0L},
/*1a02*/ {0x0L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0x0000L},
/*1a03*/ {0x0L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0x0000L},

           CO_OD_FIRST_LAST_WORD,
};
`
CANopenApp/alcosmartOD/CO_OD.c:25:10: warning: missing braces around initializer
CANopenApp/alcosmartOD/CO_OD.c:25:10: warning: (near initialization for 'CO_OD_ROM.RPDOCommunicationParameter[0]')

robincornelius, 2019-10-28:
Did you move all of the PDOs over to ROM? you cannot mix them between ram and rom and eeprom.

One thing i do notice is you have gaps in the PDO space, you are staring at index 0x1802 and not 0x1800? So either this is the issue or you have a mix of ram/rom

robincornelius, 2019-10-28:
Just checked not starting at 0x1800 is not an issue. I bet you have some in ram and some in rom.

trojanobelix, 2019-10-28:
Yes, that is right. 

If I add a TXPDO 0x180x on the TX PDO Mapping tab I can not choose which memory type is used.
I have to go to OD tab and change the memory type manually (if it isn't ROM). And not only for the communication parameter (0x180x) , but also for the mapping parameter (0x1a0x).  

Good change to miss that. And hard to find the reason of the compile errors, cause it messed up the struct braces.

So if mixing is not allowed it might be a good feature to define the memory type for PDOs global (read: chose it once, and not on every communication/mapping parameter.

robincornelius, 2019-10-28:
yea i had a quick look at the exporter and think this can be fixed, it all comes from specialarraysearch() i added a storage location to this and an outer loop to itterate around the defined locations, its almost working but some of the defines are still not correct.

trojanobelix, 2019-10-28:
That sounds good! 
Thank you, robincornelius.

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Fixed: shared records and PDO storage groups generate contiguous balanced initializers, including interleaved indices.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius