robincornelius/libedssharp #181: TPDO does not send

ArminBmrt, 2019-12-06:
I am working on controlling an epos motor controller. I have it working using PDO with python and SDO on the microcontroller. Now I want to implement PDO communication on the microcontroller.

I want to write a complex data type 64bit 0x20c1 'Interpolation data record' NodeID 2 is the consumer of this message.

When I try to send it by modifing the interpolationDataRecord  there is no message on the bus. The same way of creating the buffer and sending it using SDO worked.

Do I need to perform some additional configuration to get the PDO working?

Code snippet of sending:
`
uint8_t* buf = (uint8_t*)&CO_OD_RAM.interpolationDataRecord;

buf[0] = ((uint8_t*) &position)[0];
buf[1] = ((uint8_t*) &position)[1];
buf[2] = ((uint8_t*) &position)[2];
buf[3] = ((uint8_t*) &position)[3];
buf[4] = ((uint8_t*) &velocity)[0];
buf[5] = ((uint8_t*) &velocity)[1];
buf[6] = ((uint8_t*) &velocity)[2];
buf[7] = dt;

CO->TPDO[0]->sendRequest = true;
`

[image]

CaCtus491, 2019-12-08:
You mentioned "send it by modifing the interpolationDataRecord"

Have you enabled Change Of State monitoring on the object dictionary entry (ticking the box marked in red below)?

[image]

ArminBmrt, 2019-12-09:
I did not enabled the PDO COS. I have enabled it now. (I might have stubled upon a bug. When I save it to a file .xdd and I re open this file the checkbox is empty again. Can someone confirm this is not saved?)

Do I need to put myself in operational mode? I only put the remote node in operational mode. If so I might have to go to the canopennode forum because I got some weird errors.

[image]

robincornelius, 2019-12-10:
On Mon, 9 Dec 2019 at 13:10, ArminBmrt <notifications@github.com> wrote:

> I did not enabled the PDO COS. I have enabled it now. (I might have
> stubled upon a bug. When I save it to a file .xdd and I re open this file
> the checkbox is empty again. Can someone confirm this is [...]

I'll check out the XDD, its possible that gets lost as if there official
place for it to go, i'm sure it can be attached as an extension property to
the XML somewhere.

Yes you will need to have the nodes in operational mode for PDOs to work.
Have you got any kind of monitor for logging can messages? it makes
debugging so much easier.

Most likely weird errors are PDO configuration errors where the stack is
rejecting the configured PDOs for some reason. The next most likely error
is wrong length of PDO, where the sending and recieving ends are expectig
different lengh data packets

CaCtus491, 2019-12-11:
I can confirm that the state of the PDO COS checkbox isn't maintained after saving and reopening the XDD. I'm not sure if this is due to it not being saved, or not being loaded.

ArminBmrt, 2019-12-13:
@robincornelius With the debugger I found that some of the OD_errorStatusBits are set. This is weird because I only added this variable because canopennode depends on it. In the example provided this variable is mapped to 0x2100 which is a manufacturer specific object. For the epos2 I am using this is the 'Holding Brake Configuration'. 

But anyways this is more an issue for canopennode I think. Any help is appreciated but I will also open an issue on the canopennode page.

robincornelius, 2020-07-24:
This this can be closed, there was a bug addressed by the referenced commit above

closed 2020-07-24 by robincornelius

Cherish-forever, 2020-12-17:
@robincornelius 
PDO COS could not be save.
I have test v8.0 release libedssharp, I found that PDO COS can selected, but when I open xdd file again, the PDO COS is unselected. I do not know why.