robincornelius/libedssharp #186: Allow to open an xdc file

Reissner, 2020-02-11:
As xdc format is important one shall be able to open xdc files.

robincornelius, 2020-04-21:
I'm not sure of the technical differences between xdc and xdd, i've got the basic XML schema which i generated the XML parser from but i don't actually know if this is correct for xdc as well. 

What i really need is to read CiA-311 but i do not have access

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

The maintained WinForms editor supports XDC in its import/export paths. The shared XDD loader selects schema 1.0/1.1 from the namespace independently of the filename extension. Both underlying schema readers and device/network loading paths are covered by regressions.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius