robincornelius/libedssharp #192: LSS supported not true and related

Reissner, 2020-02-25:
In GUI it is possible to choose LSS supported. 
It is unchecked after saving but without further notification. 
This seems dangerous to me. 
Better, deactivate checkbox. 

Related: Client/Server is not part of DeviceInfo as suggests name of tab.

Reissner, 2020-02-25:
forgotten: also commented LSS_Type is no valid DeviceInfo.

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Fixed: edits mark the document dirty and save without requiring the Update button; editor regression.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius