robincornelius/libedssharp #196: Bug?: save device info

Reissner, 2020-02-26:
As far as I can see, changes at least in tab "Device Info" cannot be saved properly, 
without previously pressing button "Update". 
I dont know exactly the Background, but i think it updates just the modification time. 
But if i dont press the button, it saves silently not what is on the gui, 
but something internal, i dont know. 
The editor asks whether to overwrite and so suggests that something is written, 
but modifications i made on the GUI are not reflected in the file saved. 

All is ok if i press button 'Update'. 

I would suggest, not to suggest to save where this is not done. 

In a sense, it is clear, that the new file can be only saved with new modification time/date. 
I would suggest, to load that time/date with file, 
update whenever there is a change on the GUI 
and save with that time/date. 
In addition, that field cannot be modifiable. 
This is what is also implemented: currently, if i modify modification time/date and close file, 
i am not asked whether i want to save changes. 
So there are no changes made. 

The suggested solution makes the button "Update" superfluous. 

I think the current solution without update just reverts the file.

fenugrec, 2020-12-13:
I agree, UI is a confusing here : on other tabs it seems as soon as a modification is made the "save" button turns red, an obvious clue; the behaviour on Device info is different and not intuitive. Alternate suggestion to the one above : red Update button when modifications are made ?

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Fixed: pending Device Info edits are committed before save and survive unrelated refreshes; actual editor save/reopen regression.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius