robincornelius/libedssharp #197: Bug?: File path was removed.

Reissner, 2020-02-26:
when i load an xml file and then i do "save eds+xml+..."
a popup windows occurs with message 
   File path was removed. Please manually save as EDS once. 
This is so although tab "Device Info" shows the correct EDS path. 

If I load the eds file first without xml, and then save as eds, 
   Please manually save as XML at least once 
This seems ok, because tab "Device Info" does not show xml path. 

If i save eds first and then do again "save eds+xml+..." 
a popup windows occurs with message 
 File path was removed. Please export CO_OD.c/h once 

If i do that, next time i do "save eds+xml+..." all works fine. 

I have a nice idea: 
the eds file allows comment lines by starting with ";". 
If such a file is created through edseditor, there shall be some introductory lines like that, 
"created with edseditor version xxx" like in .h and .c files. 
Also there shall be a line with the path to save .h and .c and .xml if so. 
These comment lines shall be read by the editor. 

Suggestion: if a line starts with "; EDSeditor" a parse should be tried. 
If this does not work, that line is really ignored, else used to restore information.

trojanobelix, 2020-02-27:
I can not reproduce the "File path was removed" stuff with v0.8 2e53bbc (if the directory does exist).

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Needs exact workflow: intentional filename reset on format changes exists; reported path-loss sequence not established.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Keeping this issue open because the remaining feature, compatibility decision or reproduction requirement is not resolved by this change.