robincornelius/libedssharp #198: Creation date changes although nothing changed.

Reissner, 2020-02-26:
If i load an xml file and then save it again, 
both creation time/date and modification time/date changed. 

It should change neither. 
If something is changed, the current time/date shall be adapted 
and that is the modification date/time. 
It shall be updated on the gui and that is what shall be saved also. 

Intesting that the editor knows that there are no modifications as one can close the file 
without saving. 
so flag 'modified' must only be replaced by modification date/time. 
Caution: shall be ensured, that modification time/date is earlier than current time/date. 
Else loading not possible. 

The creation date/time shall not be affected. 
Only if saved as a new file, it shall be changed, as it is a copy. 
To that end, the path of the current file shall be taken into account. 

It is reasonable, that the current menu item is renamed into "save device file 
as" 
and that there is a menu item 'save device file' saving the current device files. 
These are all represented on the GUI.

trojanobelix, 2020-02-26:
Notepad++ just disables the "Save" entry in the file menu and the short cut until modified.
Word 2010 has obvious an internal "modified" bit: Saving does nothing until you change something.

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Unchanged XDD 1.0/1.1 round trips preserve independent creation/modification timestamps (Issue198UnchangedXddPreservesCreationAndModification). The legacy reader no longer substitutes creation time for modification time, and the 1.1 writer updates modification time only for a dirty model. EDS serialization writes the stored model timestamps rather than replacing them during an unchanged save.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius