robincornelius/libedssharp #199: OD tab: hidden button

Reissner, 2020-02-27:
in the GUI in tab "Object Dictionary", on the bottom there is the upper part of a button. 
I cannot see the label but it seems to be a kind of uptate button as pressing it turns background red. 
This is not really useable.

trojanobelix, 2020-02-27:
Are you sure you are using the latest version and have maximized the screen? Or at least not scaled too small?  This was a problem that was fixed a year ago!

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Editor now uses AutoScroll for metadata reachability; original OS/DPI-specific case still needs reproduction.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Keeping this issue open because the remaining feature, compatibility decision or reproduction requirement is not resolved by this change.