robincornelius/libedssharp #204: Question: DummyUsage

Reissner, 2020-03-03:
It seems to me that DummyUsage in eds is extracted from Section on RPDOs, 
whereas the eds editor just takes it from the loaded file. 
If my impression is true, it seems necessary also to check consistency.

robincornelius, 2020-04-21:
Dummy usage is currently incomplete, it should be generated from the RPDOs and TPDOs mapping and indicate if any dummys are used in the PDO maps, and set the variable to 1 if that data type is used as a dummy.  

Its not the most useful of features, and i don't fully understand the use case so its a pretty low priority. It could be added to the end of the PDO refactoring i need to do

trojanobelix, 2020-04-21:
For data that are not used locally can not be reflected (directly) in the OD and therefore can be substituted with a Dummy Mapping. I use it and find it very useful :-)

robincornelius, 2020-04-21:
There are two aspects to this, one is the dummy mapping which is very useful, the other is the dummy usage which is less so... unless it is expected to include DEFSTRUCT type data. Currently you can pick any of the standard variables for dummy mapping, eg INT8 INT16, UINT32 etc... these are defined in the canopen variables 1,2,3,4,5,6,7 and reflected in the [dummy usage] section in eds. All this section says is that you are using a dummy mapping of type UINT8 or UINT32 etc it does not say where. It is implict in the standard that the types for the standard variables are defined.

Where it may come in handy is if you need a dummy mapping for a non standard variable size, eg 2 bits or 4 bits. In this case its possible to define new data types and set their width using 👎 

[TypeDefinitions]
SupportedObjects=12
....
11=0x40

[40]
ParameterName=INTEGER2
ObjectType=0x5
DataType=0x7
AccessType=RO
DefaultValue=2
PDOMapping=1

this then allows you to add the following

[Dummy Usage]
Dummy0040=1

This then allows you to use the custom 2 bit data type in dummy maps.

it is also possible to say things like 

[DummyUsage]
Dummy0002=0

which says you are NOT allowed to use INT8 as a dummy map (INT8 == 0x002)

Maybe the device has a reason to not allow this in PDO maps.

trojanobelix, 2020-04-21:
Ah, okay. Missed that. but the standard @7.4.7.1 says: Other objects _(than standard  INTEGERxx/UNSEIGNEDxx)_ of the object code DEFTYPE and DEFSTRUCT shall not be mapped into RPDOs.

I think that forbids dummy mapping for a non standard variable size!?

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Enhancement: DummyUsage mapping eligibility policy requires agreement.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Keeping this issue open because the remaining feature, compatibility decision or reproduction requirement is not resolved by this change.