robincornelius/libedssharp #208: object kays not supported are grayed out, still visible

Reissner, 2020-03-03:
Consider 1018, identity, 
It is misleasing that access type is rw and memory type is ROM. 
This apparent contradition is simply resolved: the access type is not active. 

It would be nicer if there were nothing visible at all.

robincornelius, 2020-04-21:
This comes back to the fact that ROM/RAM/EEPROM are not part of CanOpen but part of the specific implementation. 

With the current can open implementation. "ROM" objects are really variables in ram that are initialised from ROM at power up. So technically they can be written to, the value is volatile but there is a defined initialisation state so will be reset every power up/reboot.

If you consider the example EEPROM code its even worse. The "ROM" values are copied from EEPROM into ram at eeprom initialisation.  It is also possible to perform a "save" function on the these variables and update their initialisation states, By using indexes 0x1010 and 0x1011 to force the save.

Any variables declared as "EEPROM" in canopennode will be written to eeprom as soon as their value changes. 

I am reluctant to change things here as its nothing specificly to do with CanOpen and only the CanOpenNode implementation and also the fact that ROM is not really ROM in all cases.

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Enhancement: disabled-property editing policy/UX needs definition.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Keeping this issue open because the remaining feature, compatibility decision or reproduction requirement is not resolved by this change.