robincornelius/libedssharp #211: EDSEditor does not handle BOOLEAN data type correctly.

grandwork2, 2020-03-20:
Each BOOLEAN data type is assigned to 1 byte instead to 1 bit in the PDOs.

This results in lots of PDO mapping errors and incorrect mapping of BOOLEAN data in the PDOs.

See as an example:

[opus.zip](https://github.com/robincornelius/libedssharp/files/4358416/opus.zip)

robincornelius, 2020-04-21:
I've found a couple of issues, i'm not currently supporting [TypeDefinitions] from EDS so no custom size objects can be used and the DEFSTRUCT does not work which is required for this EDS file attached.

Then as you point out there is no support for fields smaller that 8 bits.  I'm going to have to do a number of things to fix this. The main issue is the GUI. Its fairly easy to fix the underlying code but the GUI needs a way of displaying more entries, potentially 64. While i'm doing this i need to move some parsing logic away from the GUI so i'll look at this as part of a refactor of the PDO handling.

robincornelius, 2020-04-22:
I've made a start on this but hit a problem, the supplied eds looks like it might have an issue for example

[1600subd]
ParameterName=PDO Mapping Entry
ObjectType=0x7
AccessType=CONST
DataType=0x7
DefaultValue=0x400002
PDOMapping=0

the default value is too short so the mapping is being read incorrectly. It should be index/sub/length xxxx/yy/zz so it looks like two bytes have vanished from this and a handful of other PDO maps

robincornelius, 2020-07-23:
I believe this is now fixed in aa0e8f2d51d0214ce9df3ff7f0ce9878722bd66b
 A new PDO editor is implemented that supports single bits, and the underlying code was refactored to support bits.

closed 2020-07-23 by robincornelius