robincornelius/libedssharp #215: CANopenNode new object dictionary interface

CANopenNode, 2020-07-13:
Hello @robincornelius,

I just pushed new `301/CO_ODinterface.h`, see https://github.com/CANopenNode/CANopenNode/issues/162#issuecomment-657544754

libedssharp will need some updates. Little changes in UI and new CanOpenNodeExporter.

I managed to compile and run the tool (in Linux). I can write new CanOpenNodeExporter, at least logic. I have currently no experience with c# or Microsoft tools.

robincornelius, 2020-07-13:
> I just pushed new 301/CO_ODinterface.h, see CANopenNode/CANopenNode#162
> (comment)

Thanks!

i'll have a look at this as soon as possible.

I think i'll need to add a new exporter for the new format but i plan to
leave the old exporter there as well as I know I am still using the
original stack and CO_OD files in active projects so i will be good to have
both.

CANopenNode, 2020-07-23:
I think, it is a good idea to keep both exporters.

I would like to get rid of non-standard elements, which are not necessary any more:
  - no more custom "_project.xml" file. Use only standard XDD file. Make some changes in UI towards more compliant XDD. 
  - no more "features", no more grouping of objects into array at export, all OD objects behave individually
  - remove "disabled" attribute, instead improve adding new objects.
  - "denotation" attribute - is it usable somewhere or can it just be removed from the UI?

Improvements in adding new objects:
  - Make dialog in "Insert Profile" command - Load external file, then open a dialog, which lists available objects (indexes). User can then choose which objects to insert. User can also input an offset value, which adds an offset to all indexes.
  - Add new option in right click besides "Add New Object". Add "Duplicate Object", then ask for index.

I prepared some specification for the XDD file. It is in "CO_ODinterface.h" file. It is best viewed if html is generated by doxygen.

Some CANopenNode specific properties are added into standard elements: &lt;property name="CO_..." value="..."&gt;

A suggest some changes to current XDD format, so it could be fully compliant to standard (if "uniqueIDRef" attribute is specified in &lt;CANopenObject&gt;, then "dataType" and "accessType" attributes must not be used there). It is not a big change and old files could still be used without problems. Maybe just old OD exporter will need small update. I think, this should be a new version for libedssharp anyway.

robincornelius, 2020-07-23:
Hi Janez,

Thanks for the input. Sorry i've been a bit slow with the editor recently
since lockdown my workload has actually gone up but I have been doing small
things to the editor. I have a change i'm finishing now, to fix the PDO
editor as I discovered we had no single bit support for mapped objects (and
this is allowed) so this was quite disruptive.

On Thu, 23 Jul 2020 at 07:49, Janez <notifications@github.com> wrote:

> I think, it is a good idea to keep both exporters.
>
Hi Yes going to make it an option to choose which exporter to use. In
theory it was designed to allow multiple exporters anyway in case any other
projects (CanFestival?) needed an exporter.

> I would like to get rid of non-standard elements, which are not necessary
> any more:
>
Getting rid of as much non standard as possible is the goal

>
>    - no more custom "_project.xml" file. Use only standard XDD file. Make
>    some changes in UI towards more compliant XDD.
>
> I propose initially we disable saving in this format, but permit loading
old files,as I for one (and i am sure others) have many projects saved in
this format still.

>
>    - no more "features", no more grouping of objects into array at
>    export, all OD objects behave individually
>
> +1

>
>    - remove "disabled" attribute, instead improve adding new objects.
>
> +1

>
>    - "denotation" attribute - is it usable somewhere or can it just be
>    removed from the UI?4
>
> Ah no... this is something extra (but currently unfinished) but is part of
the standard. Denotation is part of not the eds files but the dcf files (
and in xml instead of XDD there is an XDC file). There are a number of
other fields that are related to this. The short story is a DCF file is a
save of an actual device that shows all actual parameters not just the
defaults that are in the OD. So if you have a device that can save its OD
in eeprom and you modify this in the field, you can save this state as a
DCF/XDC file to record this changes.  It is then possible to re-deploy a
device by uploading the DCF file to it. In fact i have another project
CanOpenMonitor (https://github.com/robincornelius/CanOpenMonitor) that uses
the low level libedssharp library and can already do the saving/loading of
DCF files from in field devices.

>
>
> Improvements in adding new objects:
>
>    - Make dialog in "Insert Profile" command - Load external file, then
>    open a dialog, which lists available objects (indexes). User can then
>    choose which objects to insert. User can also input an offset value, which
>    adds an offset to all indexes.
>    - Add new option in right click besides "Add New Object". Add
>    "Duplicate Object", then ask for index.
>
> Yep there are a load of changes that need doing. these are good ideas.

>
> I prepared some specification for the XDD file. It is in
> "CO_ODinterface.h" file. It is best viewed if html is generated by doxygen.
>
> Some CANopenNode specific properties are added into standard elements:
> <property name="CO_..." value="...">
>
I'll take a look as soon as i can

> A suggest some changes to current XDD format, so it could be fully
> compliant to standard (if "uniqueIDRef" attribute is specified in
> <CANopenObject>, then "dataType" and "accessType" attributes must not be
> used there). It is not a big change and old files could still be used
> without problems. Maybe just old OD exporter will need small update. I
> think, this should be a new version for libedssharp anyway.
>
Ah i had missed that. I think i can fix this with out breaking old files.
But i need to test.

Best regards
Robin

CANopenNode, 2020-07-23:
Thanks for your reply. You don't need to hurry. I have a lot of work with stack files to integrate new OD interface, anyway.

robincornelius, 2020-07-31:
I've started work on this. I've merged the outstanding pull requests to use the new 301/ etc directory structure. I have also created an IExporter interface and then created two implementations of this interface a CanOpenNodeLegacy and a CanOpenNode exporter. So all new development of the OD exporter can go in the 2nd one and leave the original intact, at least until CanOpenNode stabilises with the new OD. I've also got push triggered travis-ci builds ,so latest mono built build can always be found at https://github.com/robincornelius/libedssharp/raw/gh-pages/build/OpenEDSEditor-latest.zip

CANopenNode, 2020-08-02:
Thank you. exe file runs in Ubuntu without extra installation, just with making it executable.
As I said, I can also build it, but I have to install all mono libraries, which is not a problem.

robincornelius, 2020-08-13:
i'm still chipping away at bits, recently some more small changes to keep compatibility with the current state of the stack. I've also created a slack workspace if anyone is interested and i was hoping we could have some channels for different related projects such as the canopennode stack and my EDSEditor etc.. if anyone things this is useful please join at https://join.slack.com/t/canopennode/shared_invite/zt-ghjhfg12-qGQNWvSEmhqcSedhFaLpuA

CANopenNode, 2020-08-14:
I'm still busy with stack changes, but I will have to dig into your project a little after finishing here. I have to get some basic understanding, how it works. I'm a little old fashioned with collaborative software, but I will join slack.

One question: I don't have experience with C# and IDEs. Which (Windows) software is best to install for your project?

robincornelius, 2020-08-14:
Hi,

The quickest and easiest would just be Visual Studio Community edition,
(free) either 2017 which i use or 2019 which should be fine. This will do
C#, C,C++ etc.  It is entirely possible to use mono (and do command line),
monodevelop, eclipse etc on windows as well. But essentially with visual
studio you only need to open the solution, and hit the green play button
and it will build and run.

robincornelius, 2020-08-15:
A quick update, i have done quite a lot of the exporter. There are still some gaps and currently there is one issue stopping it compiling.

In the OD_t struct we have ->
{0x1018, 4, 0 ,ODT_REC, &OD_xyz_objs.o_1018_identity },

but in OD_xyz_objs we don't have 1018_identity but rather the children of 1018. Clearly this is wrong somewhere. But i've run out of time to work on this for today. I just need to figure out the correct way to express the child objects in REC objects, i always though the individual children had their own attributes, 

There are a few things not finished but apart from this final issue, i had everything else exporting and i managed to compile the produced .c file.

At least we have a foundation to work on now

robincornelius, 2020-08-15:
Oh i forgot to say, in the editor, go to tools->preferences and select the experimental exporter

CANopenNode, 2020-08-17:
Hi,
I installed Visual studio, joined Slack. But unfortunately this week I'm only partially available for CANopen. I have some other work to do.

I think, Slack will be good place to meet and discuss details for new OD editor. I would like to propose some small changes in principles, go through some parts of code, remove some outdated parts, ...

CANopenNode, 2020-12-01:
Latest release of libedssharp has many updates:
- CANopenNode exporter added
- Project file type updated to standard CANopen xdd, v1.1 (new importer/exporter)
- Renewed interface for Object Dictionary
- Added clone for easier insertion of new object. Updated also "insert profile"
- Renewed documentation generator for markdown
- many polishing

closed 2020-12-01 by CANopenNode