robincornelius/libedssharp #219: EDS Editor throwing an exception when parsing a specific EDS file

julian-poidevin, 2020-08-04:
Hello,

I have the following exception occuring when trying to export CanopenNode C & H from [this EDS file](https://github.com/robincornelius/libedssharp/files/5020846/mpu-25.zip).

```
System.FormatException: Les caractères supplémentaires ne pouvant pas être analysés sont à la fin de la chaîne.
   à System.ParseNumbers.StringToInt(String s, Int32 radix, Int32 flags, Int32* currPos)
   à System.Convert.ToUInt32(String value, Int32 fromBase)
   à libEDSsharp.LegacyCanOpenNodeExporter.formatvaluewithdatatype(String defaultvalue, DataType dt)
   à libEDSsharp.LegacyCanOpenNodeExporter.export_OD_def_array(String location)
   à libEDSsharp.LegacyCanOpenNodeExporter.export_c(String filename)
   à libEDSsharp.LegacyCanOpenNodeExporter.export(String folderpath, String filename, String gitVersion, EDSsharp eds)
   à ODEditor.ODEditor_MainForm.exportCanOpenNodeToolStripMenuItem_Click(Object sender, EventArgs e)
   à System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
   à System.Windows.Forms.ToolStripMenuItem.OnClick(EventArgs e)
   à System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
   à System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
   à System.Windows.Forms.ToolStripItem.FireEventInteractive(EventArgs e, ToolStripItemEventType met)
   à System.Windows.Forms.ToolStripItem.FireEvent(EventArgs e, ToolStripItemEventType met)
   à System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
   à System.Windows.Forms.ToolStripDropDown.OnMouseUp(MouseEventArgs mea)
   à System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   à System.Windows.Forms.Control.WndProc(Message& m)
   à System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   à System.Windows.Forms.ToolStrip.WndProc(Message& m)
   à System.Windows.Forms.ToolStripDropDown.WndProc(Message& m)
   à System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   à System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   à System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Assemblys chargés **************
mscorlib
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.4180.0 built by: NET48REL1LAST_B
    CodeBase : file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
EDSEditor
    Version de l'assembly : 1.0.0.0
    Version Win32 : 1.0.0.0
    CodeBase : file:///C:/Users/Julian/Desktop/Logiciels/OpenEDSEditor-latest/EDSEditor.exe
----------------------------------------
System.Windows.Forms
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.4150.0 built by: NET48REL1LAST_C
    CodeBase : file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.4001.0 built by: NET48REL1LAST_C
    CodeBase : file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.3752.0 built by: NET48REL1
    CodeBase : file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
libEDSsharp
    Version de l'assembly : 1.0.0.0
    Version Win32 : 1.0.0.0
    CodeBase : file:///C:/Users/Julian/Desktop/Logiciels/OpenEDSEditor-latest/libEDSsharp.DLL
----------------------------------------
System.Configuration
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.4190.0 built by: NET48REL1LAST_B
    CodeBase : file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.4180.0 built by: NET48REL1LAST_B
    CodeBase : file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.3752.0 built by: NET48REL1
    CodeBase : file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Windows.Forms.resources
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.3752.0 built by: NET48REL1
    CodeBase : file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_fr_b77a5c561934e089/System.Windows.Forms.resources.dll
----------------------------------------
mscorlib.resources
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.3752.0 built by: NET48REL1
    CodeBase : file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_fr_b77a5c561934e089/mscorlib.resources.dll
----------------------------------------
SourceGrid
    Version de l'assembly : 4.40.4580.29115
    Version Win32 : 4.40.4580.29115
    CodeBase : file:///C:/Users/Julian/Desktop/Logiciels/OpenEDSEditor-latest/SourceGrid.DLL
----------------------------------------
```

The code is crashing here :
https://github.com/robincornelius/libedssharp/blob/9544eba69ac841ce1de4d9fdef3977240324c4fa/libEDSsharp/LegacyCanOpenNodeExporter.cs#L1084-L1088

Here's the variables values and the call stack :
[image]

robincornelius, 2020-08-04:
The issue is (and some others like it) :-

in the eds..

[2300sub9]
ParameterName=Reset
ObjectType=7
DataType=0x0007
AccessType=wo
PDOMapping=0
DefaultValue=0.0

It says DataType is 0x007 which is an UNSIGNED32 but the DefaultValue is
given as 0.0 which is a REAL with a decimal place. It should have been just
0.

I'm not sure if its allowed to put 0.0 for this case, it might be
acceptable and i have to deal with it, or it might be an invalid EDS file
in which case, maybe i could deal with it a little more gracefully.

julian-poidevin, 2020-08-04:
Oh, I see ! I'll go ahead and try to correct the EDS.
It seems to be containing a lot of issues, I got it from a supplier.

julian-poidevin, 2020-08-04:
Problem is fixed after fixing the EDS file.
I'll let you close the issue depending if you want to implement something be more flexible with this kind of EDS bad formatting.

robincornelius, 2020-08-04:
Hi,
Its worth keeping open as you can achieve the same results inside the
editor. If you create a new object of say INT16 then set the default value
to 0.0 you get the same results as you reported. Some additional data
validation is needed

closed 2020-08-04 by robincornelius

julian-poidevin, 2020-08-05:
Wow, damn this fix was fast ! Thanks a lot !

julian-poidevin, 2020-08-05:
Works like a charm : 
[image]

Could you eventually add the index and sub-index to the warning messages so the errors can be more identifiable and fixed more easily?

Thanks a lot !

commit 7c49b150 (2020-08-04): Add a try around the value converter and log a warning (rather than crashing) if value conversion fails, fixes #219