robincornelius/libedssharp #220: Mapping Parameter typedefs do not define correct number of mapped objects

heliochronix, 2020-08-05:
The mapping parameters for PDOs (`OD_RPDOMappingParameter_t` and `OD_TPDOMappingParameter_t`) seem to base the number of mapped objects off the first PDO defined. As an example, I have RPDO 1 on a node mapping two objects, but another of my RPDOs maps three objects. The typedef is as follows:
```
*1600      */ typedef struct {
               UNSIGNED8      numberOfMappedObjects;
               UNSIGNED32     mappedObject1;
               UNSIGNED32     mappedObject2;
               }              OD_RPDOMappingParameter_t;
```
While the use here produces a compilation error because `mappedObject3` is not defined:
```
/*0x1614*/ const CO_OD_entryRecord_t OD_record1614[4] = {
           {(void*)&CO_OD_ROM.RPDOMappingParameter[20].numberOfMappedObjects, 0x0E, 0x1 },
           {(void*)&CO_OD_ROM.RPDOMappingParameter[20].mappedObject1, 0x8E, 0x4 },
           {(void*)&CO_OD_ROM.RPDOMappingParameter[20].mappedObject2, 0x8E, 0x4 },
           {(void*)&CO_OD_ROM.RPDOMappingParameter[20].mappedObject3, 0x8E, 0x4 },
};
```
```
Compiling CO_OD.c
./source/ObjDict/CO_OD.c:90:43: warning: excess elements in struct initializer
   90 | /*1614*/ {0x3L, 0x40040310L, 0x40040410L, 0x40040510L},
      |                                           ^~~~~~~~~~~
./source/ObjDict/CO_OD.c:90:43: note: (near initialization for 'CO_OD_ROM.RPDOMappingParameter[20]')
./source/ObjDict/CO_OD.c:520:56: error: 'OD_RPDOMappingParameter_t' has no member named 'mappedObject3'; did you mean 'mappedObject1'?
  520 |            {(void*)&CO_OD_ROM.RPDOMappingParameter[20].mappedObject3, 0x8E, 0x4 },
      |                                                        ^~~~~~~~~~~~~
      |                                                        mappedObject1
make: *** [../../../ChibiOS/os/common/startup/ARMCMx/compilers/GCC/mk/rules.mk:190: build/obj/CO_OD.o] Error 1
```

I believe the effort here was to minimize memory usage, but it needs to be based off the largest set of mapped objects.

robincornelius, 2020-08-12:
Arrg thats horrible,  you are correct it needs to auto generate the struct based on the largest possible set. Its worse now as well as i've added bit support to the PDO editor so you can in theory have 64 entries if all are single bits.

robincornelius, 2020-08-12:
Ok its done, try the auto build in a bit .. https://github.com/robincornelius/libedssharp/raw/gh-pages/build/OpenEDSEditor-latest.zip

But there is a use case we are missing, it may be desired to have a bigger array than currently defined, so that dynamic PDOs can be changed on the fly. This could be solved in the editor by showing as many "Empty" slots as defined as empty mappings, its all a little bit messy though

closed 2020-08-12 by robincornelius

trojanobelix, 2022-07-22:
Hello Robin,
I'm a little stumped on your solution . Maybe you have time for a quick look? https://github.com/CANopenNode/CANopenEditor/issues/31

commit 108c0583 (2020-08-12): Set the mapping typedefs to set the max number in the group they are in fixes #220