robincornelius/libedssharp #222: Duplicate PDO_MappingEntry definitions

seclorum, 2020-08-12:
Hi,

Attempting to auto-generate CO_OD.? for my project, I fire up EDSEditor v0.8 (built from the latest sources as of issue date), and export the CO_OD.? files .. but when I attempt to use the files un-edited, I get this:

`./CO_OD.h:120:31: error: duplicate member 'PDO_MappingEntry'
               UNSIGNED32     PDO_MappingEntry;
`
.. which spews on for a while, since there are many duplicate definitions ..

Is this something I'm doing wrong with the EDSEditor itself?  FWIW, I take EDS files created directly from Lenze Engineer HighLevel V2.28.0.0 without any modifications, open them in EDSEditor, and do a simple export (CANOPEN_V2) .. obviously I can fix up the files by hand myself, but it would be better (I think) to know what is happening here and whether I can fix this in the pipeline somehow ..

robincornelius, 2020-08-12:
I think this is related to #223 and is caused by the last release of 0.8 being out of date with latest development of canopennode.

Please try https://github.com/robincornelius/libedssharp/raw/gh-pages/build/OpenEDSEditor-latest.zip and remember in tools->preferences there is a option to choose the legacy exporter (0,8 version) or the new exporter which is attempting to follow the canopennode tip and will eventually be stabilised at a particular version and a new option added

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Fixed: duplicate record-member names now have distinct matching generated data pointers.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius