robincornelius/libedssharp #227: App delete unused mapped objects after click "Save" Button

adamusx, 2020-09-24:
Example:

Object 0x1A00: TPDO mapping parameter require subindex:
0 Number of mapped objects: 1
1 Mapped object 1
2 Mapped object 2
...
8 Mapped object 8

All mapped objects index 1...8 (even if only 1 mapped object is used) are required for CO_Init. 
If there is no 8 mapped objects, CO_Init() return CO_ERROR_PARAMETERS because OD_TPDOMappingParameter_t or OD_RPDOMappingParameter_t has different structure size :

` /* Verify parameters from CO_OD */
    if(   sizeof(OD_TPDOCommunicationParameter_t) != sizeof(CO_TPDOCommPar_t)
       || sizeof(OD_TPDOMappingParameter_t) != sizeof(CO_TPDOMapPar_t)
       || sizeof(OD_RPDOCommunicationParameter_t) != sizeof(CO_RPDOCommPar_t)
       || sizeof(OD_RPDOMappingParameter_t) != sizeof(CO_RPDOMapPar_t))
    {
        return CO_ERROR_PARAMETERS;
    }`

It happens after click "Save" button on Tab "TX PDO Mapping/ RX PDO Mapping". Application automatically remove unusned mapped objects.

margce, 2020-10-21:
I'm experiencing the same issue, I need to edit the output files manually so it's compatible with the canopennode libraries.

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

The merged fixes preserve physical mapping entries independently of the active mapping count and choose the full physical legacy typedef even when all PDOs are inactive. Issue268InactiveMappingsStillDefineFullPhysicalType and PDO rebuild regressions cover the structure-capacity problem. Generated C compiled against pinned CANopenNode v1.3/v4.1 headers.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius