robincornelius/libedssharp #229: Incorrect string format, but which string???

mattkefford, 2020-10-20:
I am just getting started with CANopen and trying to use your editor to create an EDS file. I am failing at the first hurdle it seems. I have typed in some information, for some reason I am unable to change the PDO amounts, but I've changed what I can and clicked update, then I get the error "Input string was not in a correct format" but it doesn't show me which string is the problem. So I am stuck...

Is there a guide on how to use this? I don't know what a valid value would be for EDS Version so I have just put the number 1, for example.

Thanks!

[image]

robincornelius, 2020-10-20:
Hi, I agree it would be helpful to know which field has failed.

In this case I think there is no hex converter attached to the Vendor
number box. So its cannot handle the hex input string.

mattkefford, 2020-10-20:
Ah yes that seems to be it 👌 now I have put 166 instead of 0xA6 , the error message does not show.

Do you have any suggestion for values I should put in "EDS Version" and the "Device commissioning" section?

closed 2020-10-20 by mattkefford

robincornelius, 2020-10-20:
Hi,

Device commissioning  you can generally leave alone. Some of these values
can be stored in the eds file, but if you are generating canopennode object
dictionaries then they are not used. They are present for completeness as
the EDS file format supports them. In theory other applications could make
use of these but i don't have anything that does.

EDS version is 4.0, the latest version of the  CiA® 306 standard i have is
1.3.0 and this states the version field shall be 4.0.
https://www.can-cia.org/can-knowledge/canopen/cia306/

But i would be surprised if anyone actually tested this or used it in anyway

Regards
Robin

mattkefford, 2020-10-20:
Ah I see, thank you that's good to know. As I said I'm new to CANopen and learning as I go.
You've been more than helpful :)