robincornelius/libedssharp #230: Is there any way to copy/duplicate objects from the directory?

margce, 2020-10-21:
I can't find an easy way to duplicate an existing object from the directory, right now I'm just editing the .xdd file manually to duplicate the entries that I need.

Any help?

CANopenNode, 2020-12-01:
New version contains clone and insert profile with interactive table to select objects and specify offset(s).

margce, 2020-12-02:
I don't see a new release version but the latest auto build has been updated, are those changes included in 0.8-99-g0425f94?

I'll try it tomorrow anyway, thanks!

CANopenNode, 2020-12-02:
yes (releases are still old)

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

The maintained WinForms editor has an object context-menu Clone action that opens the insertion/offset dialog. The merged fix also makes ODentry.Clone preserve identity, flags, compact-object metadata and custom properties, covered by Issue230ClonePreservesIdentityFlagsAndExtensions. Select the source object(s), right-click and choose Clone.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius