robincornelius/libedssharp #232: add a new RPDO

jolaizolaa, 2020-10-29:
Hi boys and girls.

Has anybody experienced an issue with the add new PDO button of the RX PDO Mapping tab? It does not add a new RPDO to the interface.

Contrary to this, the button of the TX PDO side works correctly. 

Does anybody have a solution for this?

Thank you in advance.

orthemius, 2020-12-11:
got the same issue

weblackey, 2021-06-02:
Same issue with latest version : 0.8-125-g85dfd8e

CANopenNode, 2021-06-29:
I made quite a lot of code for the latest versions. But didn't touch tabs: "TX PDO Mapping", "RX PDO Mapping" and "Modules".

"Device Info" and "Object Dictionary" are renewed and additional PDO can be added there. This is more raw approach, but I can currently maintain only that.

Eliorys, 2021-11-25:
Same issue with latest version : 0.8-125-g85dfd8e
Any information about solving this problem or a working previous version?

koson, 2022-01-04:
Hello,
On this issue, I use the cloning method in the Object Dictionary tab.
Right-click on the line "0x1403 RPDO communication parameter" and select "clone..." then edit anything you need.
The same procedure on  "0x1603 RPDO mapping parameter".

Take a look at the tab RX RPDO mapping, you'll see the addition RPDO.

I don't know how to fix this by programming.

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Fixed: PDO gap selection does not overwrite disabled existing objects.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius