robincornelius/libedssharp #233: Warning and error messages with generated CO_OD.h and CO_OD.c files using Copley EDS with CANopenNode Stack on Zephyr

clewis04, 2020-11-03:
Hello, 

I am running into issues using libedssharp on windows to generate CO_OD.h and CO_OD.c files using the Copley EDS with CANopenNode Stack on a MCU with Zephyr as the RTOS. When I try to compile with no editing of the eds and just generation of the c and h files, I get errors using the Zephyr CANopenNode Stack that I do not see when I use the socketCAN Linux with this same eds file. Specifically, the objects with no default values are causing issues. 

Has anyone run into this issue or know to resolve? Thanks, and links to eds below as a text file and as a download since .eds file types are not supposed. 

[CopleyAmp.eds.txt](https://github.com/robincornelius/libedssharp/files/5482228/CopleyAmp.eds.txt)

This was directly downloaded from Copley Website: https://www.copleycontrols.com/en/products/ap2-090-14/  Under the Downloads & Resources section under Software named CANopen EDS

edesta-be, 2020-11-09:
To add to this: it seems as though, regardless, this tool should generate valid .C and .H files that can be compiled. It appears that that is not the case. Apart from the empty commas it produces, it also suggests that there is an array that has too many variables. Any help with this would be great! Thank you!

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Fixed: missing numeric defaults, duplicate storage names and inconsistent vendor PDO member names. Supplied Copley EDS exports compile as legacy and v4 C.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius