robincornelius/libedssharp #239: CO_OD.c code including "CO_driver.h"

JuPrgn, 2020-12-10:
I have an issue with `CO_VERSION_MAJOR ` this is defined in `CO_driver.h` and also required to include `CO_driver.h` : 

```
#if CO_VERSION_MAJOR < 2
 #include "CO_driver.h"
 #include "CO_OD.h"
 #include "CO_SDO.h"
#elif CO_VERSION_MAJOR < 4
 #include "301/CO_driver.h"
 #include "CO_OD.h"
 #include "301/CO_SDOserver.h"
#else
 #error This Object dictionary is not compatible with CANopenNode v4.0 and up!
#endif
```
MPLAB X IDE run the first part `< 2` even if `CO_VERSION_MAJOR ` is equal to 2.

Should we find another way to include `CO_driver.h` /  `301/CO_driver.h` here ?

CANopenNode, 2020-12-10:
It seems, MPLAB really has a problem. It is strange.
I just don't like complicate, if only MPLAB compiler has problems. I would make a hack in CO_driver.h for MPLAB and make a comment there. For example, define CO_VERSION_MAJOR as 1. It is not used anywhere else than in CO_OD.c, anyway.

JuPrgn, 2020-12-10:
If /301 is added to compiler include path this is fine : 

```
#include "CO_driver.h"
#if CO_VERSION_MAJOR < 2
 #include "CO_OD.h"
 #include "CO_SDO.h"
#elif CO_VERSION_MAJOR < 4
 #include "CO_OD.h"
 #include "301/CO_SDOserver.h"
#else
 #error This Object dictionary is not compatible with CANopenNode v4.0 and up!
#endif
```

CANopenNode, 2020-12-10:
Of course ...
my mistake
CO_VERSION_MAJOR is defined in CO_driver.h, but #if is one line above the #include.

I think, we can't avoid using special compiler option or manual editing CO_OD.c.
- first solution is to include directory 301 into project configuration and change libedssharp as you propose.
- second solution is to add macro CO_VERSION_MAJOR=2 into the project configuration (or makefile).

I prefer second solution. However, this has to be used for all tools, not just MPLAB. libedssharo is then Ok as it is.
If we use first solutuion, we have to change also all other include directives to make things clean.

PR #240 (2020-12-23): Fix CO_VERSION_MAJOR includes
This PR fix #239
commit 097c5088 (2020-12-23): Fix CO_VERSION_MAJOR includes

This PR fix #239
closed 2020-12-23 by robincornelius