robincornelius/libedssharp #242: packaging for linux

fenugrec, 2020-12-13:
Hi,
I've just created a package for archlinux ( https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=libedssharp )
currently builds v0.8.
It basically does "make" then installs the contents of EDSTest/bin/Debug.

Next I'll be making a package to build from git; anything special I should be aware of , or may be missing ?

margce, 2020-12-15:
Hey, sorry for the off topi message but i'm planning on doing the same for macOS. Using mono I was able to compile the project but the app crashes on start, I also tried running the app with wine but I have issues with that as well. 

I guess if you end up using Wine we might have similar "bugs"

martinwag, 2020-12-22:
I've submittet a few fixes for linux some time ago and have it running with those. I'm using a version from around mid 2019 / CANopenNode 1.3, so I think you should get it working again without too much hassle.

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Platform/package request: Linux packaging is outside the maintained Windows .NET Framework 4.8 target.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Keeping this issue open because the remaining feature, compatibility decision or reproduction requirement is not resolved by this change.