robincornelius/libedssharp #247: Not possible to change number of PDOs (Linux Mono)

Qbicz, 2021-03-01:
When I open the .eds file, the number of PDOs jumps to 512 and this value is not editable ("No TX PDOs" in the screenshot, I only need 4 TPDO).
This makes using the EDSEditor very hard. also a few other fields in the PDO mapping tab are not editable and I would expect them to be.

[image]

Is this expected, or is it behaving strange only on Linux? I'm running the 0.8 release from Github with
`mono EDSEditor.exe`

Output I get:
> mono EDSEditor.exe
> Gtk-Message: 13:49:29.682: Failed to load module "canberra-gtk-module"
> libpng warning: iCCP: known incorrect sRGB profile
> libpng warning: iCCP: cHRM chunk does not match sRGB
> libpng warning: iCCP: known incorrect sRGB profile
> libpng warning: iCCP: cHRM chunk does not match sRGB
> libpng warning: iCCP: known incorrect sRGB profile
> libpng warning: iCCP: cHRM chunk does not match sRGB
> libpng warning: iCCP: known incorrect sRGB profile
> libpng warning: iCCP: cHRM chunk does not match sRGB
> libpng warning: iCCP: known incorrect sRGB profile
> libpng warning: iCCP: cHRM chunk does not match sRGB
> libpng warning: iCCP: known incorrect sRGB profile
> libpng warning: iCCP: cHRM chunk does not match sRGB
> libpng warning: iCCP: known incorrect sRGB profile
> libpng warning: iCCP: cHRM chunk does not match sRGB
> Showing combo at {Width=100, Height=16} size {X=186,Y=18}
> 

Setup:
EDSEditor version `0.8-0-gb60f4eb`
Mono version: `Mono JIT compiler version 6.8.0.105 (Debian 6.8.0.105+dfsg-2 Wed Feb 26 23:23:50 UTC 2020)`
Ubuntu 20.04.1, kernel `5.8.0-43-generic`

Qbicz, 2021-03-01:
Ok, I just found that the number of TPDO is set to 512 only if `NrOfTXPDO=0` in the `[DeviceInfo]` section. When I have set it manually (outside of the editor) to a non-zero value, the editor respects it.

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Needs reproduction: reported Mono count conversion cannot be reproduced in current Windows model/editor; Mono unavailable.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Keeping this issue open because the remaining feature, compatibility decision or reproduction requirement is not resolved by this change.