robincornelius/libedssharp #251: Data type DOMAIN doesn't work

jdascenzio, 2021-04-22:
Hi!
In Object Dictionary, when I select DOMAIN for data type, the variable generates in OD.c is like:
     dataElementSizeof = sizeof(not specified)
Thank you for any help!

CANopenNode, 2021-04-26:
Correct should be:
`dataElementSizeof = 0`

Data size for domain is application specific. I sent PR with fixes.

jdascenzio, 2021-04-26:
Thanks for your fixes!

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

The maintained v4 exporter emits callback-backed DOMAIN storage with dataElementSizeof=0 rather than sizeof(not specified). Issue251DomainArrayDoesNotEmitInvalidSizeof now explicitly covers this output. Closing the reported invalid-sizeof case as resolved in the maintained successor.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius