robincornelius/libedssharp #253: .xdd file format corrupted datatype unsigned48 to Octet String.

davidsmoot, 2021-04-29:
Cloned version today, title bar says "v0.8-126-g869ed49" and ran solution in VS 2019.

Opened an existing, known good .eds from a previous developer.  One of the SDO's has a datatype of "unsigned48" (0x0018).  

Saved the project in the .xdd format.  

The datatype of that SDO was changed from "unsigned48" to "Octet string" (0x000A).

Thanks for the tool.
David

CANopenNode, 2021-04-29:
unsigned48 and similar are deprecated in the new version. New xdd format does not support them, but older formats should still work.

davidsmoot, 2021-04-30:
Fair enough.  Closing the issue then.  I did not see a dialog to save in an older format but I suppose I could use an older version of the tool.  The solution for me was to not go through the project format.  Import EDS, make edit, export EDS.  But that leads me to another bug I am about to file...

closed 2021-04-30 by davidsmoot

JuPrgn, 2021-04-30:
The export format is configurable on Tools > Preferences