robincornelius/libedssharp #258: Unable to map Sub Indexes of Records to PDOs

StormOli, 2021-06-02:
I'm unable to map any sub index of a RECORD object to either a Tx or Rx PDO.
It either selects a random entry that doesn't even have PDO permissions, or it adds an unknown blank entry.  Sometimes it also gets stuck on the dropdown after I press enter with no way to continue without closing and opening the device. 

All other object types seem to work as expected. I'm currently running version 0.8-125-g85dfd8e

StormOli, 2021-06-04:
The issue seems to be known about.
In DevicePDOView2.cs in method Vcc_ValueChangedEvent there is a comment

//warning if the subindex is still hex the converter will not know about it
//we may need to append 0x to keep it correct

I can confirm that all my subindexes in my RECORD are marked as hex in the 2nd value of "bits".
When would this not be the case and what is the reason for not appending 0x?

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Current source contains explicit index/subindex hex parsing, but the reported random record-subindex selection and stuck dropdown have not been reproduced end to end in this review. Leaving open; please provide a minimal dictionary, selected record/subindex, editor version and platform.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Keeping this issue open because the remaining feature, compatibility decision or reproduction requirement is not resolved by this change.