robincornelius/libedssharp #259: Updating Tx PDO mapping changes access privileges incorrectly

margce, 2021-06-14:
Hello,

I'm currently using the latest auto build v0.8-126 and when updating the Tx PDO or Rx PDO Mappings the app automatically changes the access privileges to and deletes the count labels of my project. 

Here is the before and after of one of the entries as an example

`           <q1:parameter uniqueID="UID_OBJ_1800">
            <label lang="en">TPDO communication parameter</label>
            <q1:dataTypeIDRef uniqueIDRef="UID_REC_1800" />
            <q1:property name="CO_countLabel" value="TPDO" />
            <q1:property name="CO_storageGroup" value="PERSIST_COMM" />
          </q1:parameter>
          <q1:parameter uniqueID="UID_SUB_180000" access="readWrite">
            <label lang="en">max sub-index</label>
            <USINT />
            <q1:defaultValue value="6" />
          </q1:parameter>
          <q1:parameter uniqueID="UID_SUB_180001" access="readWrite">
            <label lang="en">COB-ID used by TPDO</label>
            <UDINT />
            <q1:defaultValue value="0x180" />
          </q1:parameter>
          <q1:parameter uniqueID="UID_SUB_180002" access="readWrite">
            <label lang="en">transmission type</label>
            <USINT />
            <q1:defaultValue value="254" />
          </q1:parameter>
          <q1:parameter uniqueID="UID_SUB_180003" access="readWrite">
            <label lang="en">inhibit time</label>
            <UINT />
            <q1:defaultValue value="100" />
          </q1:parameter>
          <q1:parameter uniqueID="UID_SUB_180004" access="readWrite">
            <label lang="en">compatibility entry</label>
            <USINT />
            <q1:defaultValue value="0" />
          </q1:parameter>
          <q1:parameter uniqueID="UID_SUB_180005" access="readWrite">
            <label lang="en">event timer</label>
            <UINT />
            <q1:defaultValue value="0" />
          </q1:parameter>
          <q1:parameter uniqueID="UID_SUB_180006" access="readWrite">
            <label lang="en">SYNC start value</label>
            <USINT />
            <q1:defaultValue value="0" />
          </q1:parameter>`

After accessing the Tx PDO mapping and saving changes this happens to the TPDO info, note how the count label gets erased and the access for every field has changed to noAccess, the same happens to TPDO mapping parameters items on 0x1A00...

`          <q1:parameter uniqueID="UID_OBJ_1800">
            <label lang="en">TPDO communication parameter</label>
            <q1:dataTypeIDRef uniqueIDRef="UID_REC_1800" />
															 
            <q1:property name="CO_storageGroup" value="PERSIST_COMM" />
          </q1:parameter>
          <q1:parameter uniqueID="UID_SUB_180000" access="noAccess">
            <label lang="en">max sub-index</label>
            <USINT />
            <q1:defaultValue value="6" />
          </q1:parameter>
          <q1:parameter uniqueID="UID_SUB_180001" access="noAccess">
            <label lang="en">COB-ID used by TPDO</label>
            <UDINT />
            <q1:defaultValue value="0x180" />
          </q1:parameter>
          <q1:parameter uniqueID="UID_SUB_180002" access="noAccess">
            <label lang="en">transmission type</label>
            <USINT />
            <q1:defaultValue value="254" />
          </q1:parameter>
          <q1:parameter uniqueID="UID_SUB_180003" access="noAccess">
            <label lang="en">inhibit time</label>
            <UINT />
            <q1:defaultValue value="100" />
          </q1:parameter>
          <q1:parameter uniqueID="UID_SUB_180004" access="noAccess">
            <label lang="en">compatibility entry</label>
            <USINT />
            <q1:defaultValue value="0" />
          </q1:parameter>
          <q1:parameter uniqueID="UID_SUB_180005" access="noAccess">
            <label lang="en">event timer</label>
            <UINT />
            <q1:defaultValue value="0" />
          </q1:parameter>
          <q1:parameter uniqueID="UID_SUB_180006" access="noAccess">
            <label lang="en">SYNC start value</label>
            <USINT />
            <q1:defaultValue value="0" />
          </q1:parameter>`

StormOli, 2021-06-25:
I have the same issues when editing PDOs.  I've created a pull request that appears to fix the issue.

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Fixed: PDO rebuild preserves communication/mapping metadata, disabled objects and the last slot. Legacy and v4 regressions.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius