robincornelius/libedssharp #265: PDOHelper.cs:411 misprint

orthemius, 2022-06-27:
there is
`if( (configindex<0x1400) || (configindex >= 0x1a00)  || ((configindex>=1600) && (configindex<0x1800)))`
1600 in dec
must be in hex:
`if( (configindex<0x1400) || (configindex >= 0x1a00)  || ((configindex>=0x1600) && (configindex<0x1800)))`

trojanobelix, 2022-06-28:
Fixed with pull request #266

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

The PDO index range check already uses hexadecimal 0x1600 in the maintained implementation. The new PDO rebuild/index regressions also pass. Closing the obsolete decimal-literal report as resolved in the maintained successor.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius