robincornelius/libedssharp #271: error on openning .eds file from ROS2-canopen

wei1224hf, 2024-05-05:
here is the error:

`System.Collections.Generic.KeyNotFoundException: 给定关键字不在字典中。
   在 System.Collections.Generic.Dictionary`2.get_Item(TKey key)
   在 libEDSsharp.EDSsharp.Loadfile(String filename)
   在 ODEditor.ODEditor_MainForm.openEDSfile(String path, Filetype ft)
`

when trying to open ros2_canopen/canopen_tests/config/cia402_system/cia402_slave.eds

ZeroErrControl, 2025-01-25:
You can refer to [this project:https://github.com/ZeroErrControl/eRob_CANopen_Python](https://github.com/ZeroErrControl/eRob_CANopen_Python).

trojanobelix, 2025-01-27:
Here is the current repro of the CANopenEditor, this is no longer maintained 

https://github.com/CANopenNode/CANopenEditor

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Fixed: missing FileInfo imports with defaults and a warning; invalid mandatory sections give a clear error. ROS2 attachment imports 70 objects.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius